purduesigbots/pros-cli #181: Device never ACK'd to b'\x08\x00\x00\x00'

mattfbacon, 2021-11-16:
#### Expected Behavior:
I expected the program to download

#### Actual Behavior:
An error message was printed.

#### Steps to reproduce:
Connect the joystick to the microcontroller via USB, and connect the computer to the joystick using the programming module.

#### System information:
Operating System:  Arch Linux

PROS Version: `pros, version 3.3.2`

#### Additional Information
The error message is always the same.

#### Screenshots/Output Dumps/Stack Traces
`pros upload --target cortex /dev/ttyUSB0`
```
Uploading  output.bin to Cortex on /dev/ttyUSB0
  Tether: Serial w/ a USB Cable
  Cortex: F/W 4.25 w/ 7.61 V (Backup: 0.00 V)
Joystick: F/W 4.25 w/ 7.02 V
Uploading /home/matt/Desktop/code/pros/first/bin/output.bin [---------------
---------------------]   0%
ERROR - pros.cli.upload:upload - Device never ACK'd to b'\x08\x00\x00\x00'
b'\x08\x00\x00\x00'
    raise VEXCommError(f"Device never ACK'd to {command}", command)
pros.serial.devices.vex.comm_error.VEXCommError: Device never ACK'd to b'\x0
8\x00\x00\x00'
b'\x08\x00\x00\x00'
We detected something went wrong! Do you want to send a report? [y/N]: n
Not sending bug report.
```

HotelCalifornia, 2021-11-16:
it's been a very long time since any of us have worked with the cortex. can you post the output of `pros lsusb`

mattfbacon, 2021-11-16:
If if I had any choice I wouldn't be using Cortex. Anyway...

`pros lsusb`
```
There are no connected VEX EDR V5 System Ports
There are no connected VEX EDR V5 User ports
VEX EDR Cortex Microcontroller Ports:
/dev/ttyUSB0 - USB-Serial Controller
```

mattfbacon, 2021-11-16:
Also thanks for the quick response.

mattfbacon, 2021-11-21:
Any insight here? I'd love to use PROS if I can.

nathan-moore, 2021-11-21:
>  I'd love to use PROS if I can.

Can other programs upload to your cortex?

mattfbacon, 2021-11-21:
I'm able to download via RobotMesh Studio (which is what we're currently using). Interestingly, downloads over the USB A-to-A directly connected to the cortex do work with RMS, despite the fact that A-to-A download is broken and/or not implemented in PROS on Linux.

I was also not able to download with PROS on Windows. Can't remember the exact error ATM though.

mattfbacon, 2021-11-21:
> What version of the cli are you running on? 3.3.2 seems to be the kernel version.

3.2.2 for the CLI. How would I check the kernel version?

nathan-moore, 2021-11-23:
The kernel version shouldn't matter. Do you mind giving an earlier release a try? https://github.com/purduesigbots/pros-cli/releases/tag/3.1.0

mattfbacon, 2021-11-23:
Will do next time I get a chance.

mattfbacon, 2021-12-10:
Now I get the error: "Device never ACK'd to 49".

mattfbacon, 2021-12-10:
Can you give me the specific command I should run? I might be downloading it incorrectly.