purduesigbots/pros-cli #192: Vscode PROS executable location

bwees, 2022-01-19:
Where is the PROS-cli executable installed when you download and use the vscode extension? My computer has a weird setup for setting the PATH (school laptop so they have to add for me) so I can't use `pros lsusb` or whatever command from inside vscode command line. If I know where the executable is I can have them add to the PATH.

Thanks

BennyBot, 2022-01-19:
Depends on your operating system.
The cli needs to go into PATH, and the toolchain needs to go into a variable called PROS_TOOLCHAIN

For Windows, the paths are as followed:
Pros CLI: %appdata%\Code\User\globalStorage\sigbots.pros\install\pros-cli-windows\
Pros toolchain: %appdata%\Code\User\globalStorage\sigbots.pros\install\pros-toolchain-windows\usr\

For Linux, the paths should be as followed:
Pros CLI: $HOME/.config/Code/User/globalStorage/sigbots.pros/install/pros-cli-linux/
Pros toolchain: $HOME/.config/Code/User/globalStorage/sigbots.pros/install/pros-toolchain-linux/usr/

For macOS, the paths should be as followed:
Pros CLI: $HOME/Library/Application Support/Code/User/globalStorage/sigbots.pros/install/pros-cli-macos/
Pros toolchain: $HOME/Library/Application Support/Code/User/globalStorage/sigbots.pros/install/pros-toolchain-macos/usr/

I can't guarantee the exact path for Linux and macOS as I'm on a Windows machine right now, but I'm pretty sure those are correct.

bwees, 2022-01-19:
Awesome Thank You!

closed 2022-01-19 by bwees