purduesigbots/pros-cli #198: Unable to upload non-pros .bin file to v5

superrm11, 2022-02-09:
I'm working on getting a VSCode version of VEXCode Pro working on linux. Building the project works, and produces a .bin file, but attempting to upload it is throwing python errors. The .ini file uploads fine.

I tried it with a normal empty pros project as well, and it uploaded.

The command:
`prosv5 upload --target v5 --project build/2021-2022-BIG.bin --slot 1 --name 2021-2022-BIG --icon USER926x.bmp /dev/ttyACM0`

Output:
```
Uploading program "2021-2022-BIG" (ttyACM0) to V5 slot 1 on /dev/ttyACM0
Uploading slot_1.ini [####################################] 100%
ERROR - pros.cli.upload:upload - File or stream is not seekable.
  File "/home/vsts/work/1/s/pros/serial/devices/vex/v5_device.py", line 339, in write_program
  File "/home/vsts/work/1/s/pros/serial/devices/vex/v5_device.py", line 535, in write_file
io.UnsupportedOperation: File or stream is not seekable.
We detected something went wrong! Do you want to send a report? [y/N]:  
Not sending bug report.
Sentry is attempting to send 1 pending error messages
Waiting up to 2 seconds
Press Ctrl-C to quit
```

Additional Info:
OS: EndeavorOS 
Pros version: 3.2.3
Python version: 3.10.2
Project: [https://github.com/RIT-VEX-U/2021-2022-BIG/](https://github.com/RIT-VEX-U/2021-2022-BIG/)
Make command: `make T=/windows/Program\ Files\ \(x86\)/VEX\ Robotics/VEXcode\ Pro\ V5/sdk`

superrm11, 2022-02-09:
```
INFO - pros.cli.common:callback - Debugging messages enabled
DEBUG - pros:callback - CLI Version: 3.2.3
DEBUG - pros.serial.ports:list_all_comports - Connected: {'device': '/dev/ttyACM1', 'name': 'ttyACM1', 'description': 'VEX Robotics V5 Brain - A9EC2700 - VEX Robotics User Port', 'hwid': 'USB VID:PID=2888:0501 LOCATION=1-3:1.2', 'vid': 10376, 'pid': 1281, 'serial_number': None, 'location': '1-3:1.2', 'manufacturer': 'VEX Robotics, Inc', 'product': 'VEX Robotics V5 Brain - A9EC2700', 'interface': 'VEX Robotics User Port', 'usb_device_path': '/sys/devices/pci0000:00/0000:00:14.0/usb1/1-3', 'device_path': '/sys/devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3:1.2', 'subsystem': 'usb', 'usb_interface_path': '/sys/devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3:1.2'};{'device': '/dev/ttyACM0', 'name': 'ttyACM0', 'description': 'VEX Robotics V5 Brain - A9EC2700 - VEX Robotics Communications Port', 'hwid': 'USB VID:PID=2888:0501 LOCATION=1-3:1.0', 'vid': 10376, 'pid': 1281, 'serial_number': None, 'location': '1-3:1.0', 'manufacturer': 'VEX Robotics, Inc', 'product': 'VEX Robotics V5 Brain - A9EC2700', 'interface': 'VEX Robotics Communications Port', 'usb_device_path': '/sys/devices/pci0000:00/0000:00:14.0/usb1/1-3', 'device_path': '/sys/devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3:1.0', 'subsystem': 'usb', 'usb_interface_path': '/sys/devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3:1.0'}
INFO - pros.cli.common:resolve_v5_port - Automatically selected /dev/ttyACM0
DEBUG - pros.cli.upload:upload - Arguments: {'target': 'v5', 'slot': 0, 'remote_name': '2021-2022-BIG', 'icon': 'USER926x.bmp', 'run_after': <FTCompleteOptions.RUN_SCREEN: 3>, 'quirk': 0, 'program_version': None, 'ini_config': None, 'compress_bin': True}
DEBUG - pros.serial.ports.direct_port:create_serial_port - Opening serial port /dev/ttyACM0
DEBUG - pros.serial.devices.vex.v5_device:query_system_version - Sending simple 0xA408 command
DEBUG - pros.serial.devices.vex.vex_device:_tx_packet - V5Device TX: C9 36 B8 47 A4
DEBUG - pros.serial.devices.vex.vex_device:_txrx_packet - TX:C9 36 B8 47 A4	RX:AA 55 A4 08 01 01 00 00 00 10 03 00
DEBUG - pros.serial.devices.vex.v5_device:query_system_version - Completed simple 0xA408 command
Uploading program "2021-2022-BIG" (ttyACM0) to V5 slot 1 on /dev/ttyACM0 (compressed)
DEBUG - pros.serial.devices.vex.v5_device:get_system_status - Sending ext 0x22 command
DEBUG - pros.serial.devices.vex.v5_device:query_system_version - Sending simple 0xA408 command
DEBUG - pros.serial.devices.vex.vex_device:_tx_packet - V5Device TX: C9 36 B8 47 A4
DEBUG - pros.serial.devices.vex.vex_device:_txrx_packet - TX:C9 36 B8 47 A4	RX:AA 55 A4 08 01 01 00 00 00 10 03 00
DEBUG - pros.serial.devices.vex.v5_device:query_system_version - Completed simple 0xA408 command
DEBUG - pros.serial.devices.vex.vex_device:_tx_packet - V5Device TX: C9 36 B8 47 56 22 00 60 FC
DEBUG - pros.serial.devices.vex.vex_device:_txrx_packet - TX:C9 36 B8 47 56 22 00 60 FC	RX:AA 55 56 29 22 76 00 01 01 00 00 01 01 00 01 01 01 00 08 06 00 00 00 00 27 EC A9 00 03 00 00 21 00 00 00 01 00 20 00 01 01 00 07 54 A1
DEBUG - pros.serial.devices.vex.v5_device:_rx_ext_packet - Set msg window to 00 01 01 00 00 01 01 00 01 01 01 00 08 06 00 00 00 00 27 EC A9 00 03 00 00 21 00 00 00 01 00 20 00 01 01 00 07
DEBUG - pros.serial.devices.vex.v5_device:_txrx_ext_struct - Unpacking with format: <x12B3xBI12xB3x
DEBUG - pros.serial.devices.vex.v5_device:get_system_status - Completed ext 0x22 command
INFO - pros.serial.devices.vex.v5_device:generate_ini_file - Created ini: [program]
version = 0.0.0
name = 2021-2022-BIG
slot = 0
icon = USER926x.bmp
description = Created with PROS
date = 2022-02-09T10:33:42.864408

INFO - pros.serial.devices.vex.v5_device:write_program - Created ini: [program]
version = 0.0.0
name = 2021-2022-BIG
slot = 0
icon = USER926x.bmp
description = Created with PROS
date = 2022-02-09T10:33:42.864408

DEBUG - pros.serial.devices.vex.v5_device:execute_program_file - Sending ext 0x18 command
DEBUG - pros.serial.devices.vex.v5_device:execute_program_file - VID: 1	Options: 128	File name: 	Run: False
DEBUG - pros.serial.devices.vex.vex_device:_tx_packet - V5Device TX: C9 36 B8 47 56 18 1A 01 80 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 A6 C1
DEBUG - pros.serial.devices.vex.vex_device:_txrx_packet - TX:C9 36 B8 47 56 18 1A 01 80 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 A6 C1	RX:AA 55 56 04 18 76 4B A6
DEBUG - pros.serial.devices.vex.v5_device:execute_program_file - Completed ext 0x18 command
DEBUG - pros.serial.devices.vex.v5_device:query_system_version - Sending simple 0xA408 command
DEBUG - pros.serial.devices.vex.vex_device:_tx_packet - V5Device TX: C9 36 B8 47 A4
DEBUG - pros.serial.devices.vex.vex_device:_txrx_packet - TX:C9 36 B8 47 A4	RX:AA 55 A4 08 01 01 00 00 00 10 03 00
DEBUG - pros.serial.devices.vex.v5_device:query_system_version - Completed simple 0xA408 command
INFO - pros.serial.devices.vex.v5_device:write_file - Transferring slot_1.ini (143 bytes) to the V5 from <_io.BytesIO object at 0x7fe233bf7360>
DEBUG - pros.serial.devices.vex.v5_device:ft_initialize - Sending ext 0x11 command
DEBUG - pros.serial.devices.vex.v5_device:ft_initialize - Initializing file transfer w/: {'function': 1, 'target': 1, 'vid': 1, 'overwrite': True, 'options': 1, 'length': 143, 'addr': 58720256, 'crc': 929607557, 'type': b'ini', 'timestamp': 697718022, 'version': 16777216, 'name': b'slot_1.ini'}
[... cut ...]
  File "/home/vsts/work/1/s/pros/serial/devices/vex/v5_device.py", line 535, in write_file
io.UnsupportedOperation: File or stream is not seekable.
We detected something went wrong! Do you want to send a report? [y/N]: 
Not sending bug report.
DEBUG - urllib3.connectionpool:_new_conn - Starting new HTTPS connection (1): sentry.io:443
Sentry is attempting to send 1 pending error messages
Waiting up to 2 seconds
Press Ctrl-C to quit
DEBUG - urllib3.connectionpool:_make_request - https://sentry.io:443 "POST /api/1226033/envelope/ HTTP/1.1" 200 2
```

HotelCalifornia, 2022-02-09:
~~incidentally the reason it's not complaining that it couldn't find a pros project here even though there's none in the directory you've given it is because the upload command does not require a project flag and allows project to be None.~~

actually, i take that back. it's got to be interpreting the serial port as the path to a binary. so it successfully opens the serial port as a file and then fails because that's a non-seekable file (obviously)

HotelCalifornia, 2022-02-09:
i'm honestly not sure what checks we could possibly add to mitigate this issue....... but i think i'm content in saying it's such a fringe case we can just ignore it most of the time

HotelCalifornia, 2022-02-09:
gonna close this now. feel free to reopen if that doesn't work