purduesigbots/pros-cli #205: CLI >3.2.3 produces ModuleNotFoundError inside virtualenvs

bnhrkn, 2022-03-07:
#### Expected Behavior:
Run `pros --version` with pros-cli installed in a virtualenv, and receive the version.

#### Actual Behavior:
Run `pros --version` with pros-cli installed in a virtualenv produces module not found error, see below.

#### Steps to reproduce:
Execute:
```
python -m venv .pros-cli
source .pros-cli/bin/activate
pip install pros-cli
pros --version
```

#### System information:
Operating System:  Reproduced on Void Linux and NixOS

PROS Version: N/A?

#### Additional Information
I don't know if the virtualenv is consequential but I assume it is because I didn't see any issues regarding normal installation.

In a virtualenv there is no error with the following:
```
python -m venv /tmp/test-venv
source /tmp/test-venv/bin/activate
pip install pros-cli==3.2.3
pros-cli --version
```
it returns `pros, version 3.2.3` as you would expect.

#### Screenshots/Output Dumps/Stack Traces
```
Traceback (most recent call last):
  File "/home/ben/.pros-cli/bin/pros", line 5, in <module>
    from pros.cli.main import main
  File "/home/ben/.pros-cli/lib/python3.9/site-packages/pros/cli/main.py", line 15, in <module>
    from pros.cli.common import default_options, root_commands
  File "/home/ben/.pros-cli/lib/python3.9/site-packages/pros/cli/common.py", line 4, in <module>
    from pros.ga.analytics import analytics
ModuleNotFoundError: No module named 'pros.ga'
```

HotelCalifornia, 2022-03-07:
will be fixed in #202

BennyBot, 2022-03-07:
We removed the latest release from PyPi, so just doing `pip install pros-cli` should work fine now.

closed 2022-03-07 by BennyBot