purduesigbots/pros-cli #223: [HELP?] Set port in `project.pros`

karmanyaahm, 2022-08-20:
#### Expected Behavior:
There are options to set the system and user ports in the project file to custom (not autodetected) locations. I want to set them to virtual (./virtualcom0 or 1) non-USB ports.

#### Actual Behavior:
There don't seem to be any documented options? I looked through the code and couldn't find anything?

#### System information:
Operating System:  Arch Linux
PROS Version: v3.3.3 from git

BennyBot, 2022-08-20:
For now, you can use a specific port passing an option when you call the upload command.
`pros upload —port="specific-port"`
You should be able to see which ports are connected to a brain through the command `pros ls-usb`.
Let me know if that doesn’t work, because I am unfamiliar with the behavior through virtual ports. I don’t see why it wouldn’t work though.

I will look at getting this added to the upload_options in project.pros, but it may take some time as we are busy with VSCode updates.

karmanyaahm, 2022-08-20:
`pros lsusb` doesn't work on virtual serial ports*. Though `pros upload --port=` works, thanks. However, it's extra typing every time (or I'll have to make aliases).

Another workaround I'm using right now is adding this
```py
ports += [ListPortInfo('./virtualVEXUser', skip_link_detection=True), ListPortInfo('./virtualVEXSystem',skip_link_detection=True)]
```
to
https://github.com/purduesigbots/pros-cli/blob/f57a010c5f6dea18f2860de878c44f363bce677c/pros/serial/ports/__init__.py#L11-L15

*I'm making these using socat with `socat -d -d TCP:robo-orangepi:10000,forever pty,link=./virtualVEXSystem,wait-slave` (I'll publish more info once I finish my project)

BennyBot, 2022-08-20:
Cool. I’ll close this once we get ports added to upload_options.

ayushuk, 2023-10-05:
Completed in #292

closed 2023-10-05 by ayushuk