purduesigbots/pros-cli #228: [FEATURE] Update VexOS

SizzinSeal, 2022-09-23:
#### Expected Behavior:
There should be a command to update VexOS.

#### Actual Behavior:
Unlike Vexcode (VSC extension or standalone), the PROS CLI does not have a command to update the firmware. The user has to download Vexcode if they want to update the firmware.

BennyBot, 2022-09-23:
This is currently not possible afaik. Updating vexOS is not simply running a command on the brain, we would have to be able to fetch the vexOS releases from VEX (which currently is not). This is not to say it will never be possible within PROS. We are going to incorporate VEX's vexcom binary into our CLI bundle so that we have access to their exclusive commands, such as updating firmware and running the battery medic program.

omegaStag, 2022-10-04:
what ben said

omegaStag, 2022-10-04:
^^ what ben said

closed 2022-10-04 by omegaStag