purduesigbots/pros-cli #229: [FEATURE] Feature Request Template

SizzinSeal, 2022-09-23:
#### Expected Behavior:
When opening an issue, there should be an option to open an issue template.

#### Actual Behavior:
There is no feature request template.

#### Additional Information
The addition of a feature request template will make it easier for inexperienced people to create quality feature requests, and will also make organization for repo managers easier.

BennyBot, 2022-09-23:
Have you taken a look at the vsc templates ([VSC templates](https://github.com/purduesigbots/pros-vsc/issues/new/choose))? I was planning on moving these over to the pros-cli and pros repos soon, just haven't gotten around to it yet. Any suggested changes to these VSC templates?

SizzinSeal, 2022-09-23:
No, I think it will work well. I'll open a PR that implements them

PR #230 (2022-09-23): add FEATURE_REQUEST.md
#### Summary:
Adds a feature request issue template. Its just the same as the one in [VSC](https://github.com/purduesigbots/pros-vsc/blob/develop/.github/ISSUE_TEMPLATE/feature-request-template.md). All credit to @BennyBot and other contributors.

#### Motivation:
Inexperienced users may not open quality feature requests. A feature request template will ensure that they provide all the necessary information

##### References (optional):
closes #229
closed 2022-09-23 by BennyBot