purduesigbots/pros-cli #232: 🐛Workflows never finishes

SizzinSeal, 2022-09-28:
#### Expected Behavior:
For the workflows to stop after a few minutes.

#### Actual Behavior:
The workflow gets stuck when building for macOS ARM

#### Steps to reproduce:
Commit and push to the repo

#### Additional Information
GitHub workflow time is billed (IIRC). It could cost a lot of money letting it run like this

#### Screenshots/Output Dumps/Stack Traces
[image]

omegaStag, 2022-10-04:
Good catch, @kunwarsahni01 pls help

kunwarsahni01, 2022-10-05:
Ah, a couple things here, we actually use Azure Pipelines not github workflows (Azure is free for us no matter what), we also have a built in timeout in all pipelines so no pipeline will be running for more than 15 minutes. And in this case it was just a case of the M1 in the cloud we are running getting turned off, so restarted it and it should be all good.

closed 2022-10-05 by kunwarsahni01