purduesigbots/pros-cli #238: Generated compile_commands.json can contain invalid arguments🐛

bnhrkn, 2022-10-08:
#### Expected Behavior:

When `WARNFLAGS+= -fanalyzer` is present in `Makefile` and `pros build-compile-commands all` is executed, the `arguments` array in `compile_commands.json` should not contain `"-fanalyzer"`.

#### Actual Behavior:
`arguments` array in `compile_commands.json` contained `"-fanalyzer"`.

#### Steps to reproduce:

1. New pros project
2. Modify makefile to contain `WARNFLAGS+= -fanalyzer`
3. Execute `pros make-compile-commands all`

#### System information:
Operating System:  Linux, Fedora 36

PROS Version: version 3.3.3

#### Additional Information
When using an IDE with clangd, an error `Unknown argument '-fanalyzer'; did you mean '-Xanalyzer'?clang(drv_unknown_argument_with_suggestion)` is produced on every file.

Andrewtho907, 2022-10-25:
By PROS version do you mean Kernel or CLI?

omegaStag, 2022-10-25:
`-fanalyzer` isn't supported as a flag with the version of arm-none-eabi-g++ that is built into the CLI.

closed 2022-10-25 by omegaStag