purduesigbots/pros-cli #248: 🐛Port Detection Isses

BennyBot, 2022-11-22:
#### Expected Behavior:
If I plug in a brain or controller to the computer, it should be consistently detecting it.
If only one VEX device is plugged in, it should automatically select that device.
If multiple devices are plugged in, it should display a list of ports that are connected and let the user choose between them.

#### Actual Behavior:
Brain and controllers are often not detected through ports. Something is wrong with the way we detect if a port is a vex device or not. I have seen examples where irrelevant arguments (such as --slot) have affected whether or not the brain is detected by the CLI or not.
If a user is connected to ONE brain or controller through a USB hub, the CLI will still prompt them for a port even though only one device is connected
If multiple devices are connected, the prompt displayed is really strange. Sometimes it has the user only able to choose one port. Sometimes the user has no ports to choose. Sometimes it displays the same port multiple times. 

It is just really inconsistent, and something is almost certainly wrong with it.

#### Steps to reproduce:
Just play around with the upload command or ls-usb and you will encounter strange behavior

#### System information:
All platforms. CLI <= 3.4.1

Jackman3323, 2023-10-04:
VTOW reported a problem with detecting the brain:

**Person:** Bry | MSOE1 | 5062F

**System Info:** Thinkpad w/ intel CPU running windows. Accessing pros cli via a wsl2 (basically a linux vm inside windows) with the Ubuntu distro of linux. Within this vm, running lsusb lists the brain as expected, whereas running pros lsusb returns a message saying no vex stuff was found:

`root@AAD-PF4DYDH1:/workspaces/over-under# root@AAD-PF4DYDH1:/workspaces/over-under# lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux 5.15.90.1-microsoft-standard-WSL2 vhci_hcd USB/IP Virtual Host Controller
Bus 001 Device 005: ID 2888:0501 VEX Robotics, Inc VEX Robotics V5 Brain - CA034800
Bus 001 Device 001: ID 1d6b:0002 Linux 5.15.90.1-microsoft-standard-WSL2 vhci_hcd USB/IP Virtual Host Controller
root@AAD-PF4DYDH1:/workspaces/over-under# pros lsusb
There are no connected VEX EDR V5 System Ports
There are no connected VEX EDR V5 User ports
There are no connected VEX EDR Cortex Microcontroller Ports`

nickmertin, 2024-04-05:
I am seeing similar behaviour as the above comment, running on Pop_OS linux (no VM):
[image]