purduesigbots/pros-cli #294: ✨Template Makefile Patching

SizzinSeal, 2023-07-27:
#### Requested Feature
Add functionality for templates to patch the makefile and compile_commands.json

#### Is this Feature Related to a Problem?
No

#### Benefits of Feature
Templates can implement more advanced features which requires the makefile to be changed. l

#### Additional Information
Regex could be used to patch specific lines, and also prevents compatibility issues it multiple templates are used.

BennyBot, 2023-07-27:
Can you elaborate on this" which parts of the makefile would a template need to change? If two templates modify the same part differently, how is this expected to work? Are we to assume that the user will not use two templates that both need to modify a certain part of the makefile. 

This is definitely something to look at for branchline @ayushuk. Handling templates that have their own separate side programs,  etc. is something that will inevitably need to be addressed. I can't remember if I have done this yet or not, but the default Makefile in kernel needs to be modified to reflect this: https://www.vexforum.com/t/adding-a-static-library-to-pros/115049/2?u=bennyboy, that way templates stop sometimes including the pros headers in their zip file

SizzinSeal, 2023-07-29:
Here's an example:

You want to display a GIF on the brain screen. You have a few options:

1. Load from a micro-sd card
2. Store as a massive array in a header somewhere
3. Upload the image as part of the upload binary

The third option would require the makefile to be modified.

SizzinSeal, 2023-07-29:
As for what parts of the makefile should be able to be edited:

- Macros
- Build targets
- Default target
- Compiler flags

And of course new text should be able to be added as well.

If 2 templates overwrite the same part of the makefile (e.g the same macro) the conductor should throw an error when attempting to apply the second template. Its up to the template developers to ensure compatibility.

SizzinSeal, 2024-04-16:
closing, as I've found a workaround:

any `.mk` file in the `firwmare` folder will be included in the makefile, making this functionality possible

closed 2024-04-16 by SizzinSeal