purduesigbots/pros-cli #295: SSL issues

jman13378, 2023-08-30:
#### Expected Behavior:
connect and updated

#### Actual Behavior:
it errored

#### Steps to reproduce:
Not sure i just installed pros

#### System information:
Platform: V5
Name      Version    Origin
--------  ---------  -------------
kernel    3.8.0      pros-mainline
okapilib  4.8.0      pros-mainline

#### Additional Information

#### Screenshots/Output Dumps/Stack Traces
```diff
PS C:\Users\1300286\over-under> pros conduct upgrade
Upgrading kernel
Updating pros-mainline... ERROR - __main__:main - HTTPSConnectionPool(host='purduesigbots.github.io', port=443): Max retries exceeded with url: /pros-mainline/pros-mainline.json (Caused by SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:1129)'))) - pros-cli version:3.4.3
  File "requests\sessions.py", line 703, in send
  File "requests\adapters.py", line 517, in send
requests.exceptions.SSLError: HTTPSConnectionPool(host='purduesigbots.github.io', port=443): Max retries exceeded with url: /pros-mainline/pros-mainline.json (Caused by SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:1129)')))
We detected something went wrong! Do you want to send a report? [y/N]: y
Sending bug report.
Want to get updates? Visit https://pros.cs.purdue.edu/report.html?event=980c477312d94b0388cd27e7ac0195ae
WARNING - urllib3.connectionpool:urlopen - Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:1129)'))': /api/1226033/store/ - pros-cli version:3.4.3
Sentry is attempting to send 2 pending events
Waiting up to 2 seconds
Press Ctrl-Break to quit
WARNING - urllib3.connectionpool:urlopen - Retrying (Retry(total=1, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:1129)'))': /api/1226033/store/ - pros-cli version:3.4.3
WARNING - urllib3.connectionpool:urlopen - Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:1129)'))': /api/1226033/store/ - pros-cli version:3.4.3
WARNING - urllib3.connectionpool:urlopen - Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:1129)'))': /api/1226033/envelope/ - pros-cli version:3.4.3
WARNING - urllib3.connectionpool:urlopen - Retrying (Retry(total=1, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:1129)'))': /api/1226033/envelope/ - pros-cli version:3.4.3
WARNING - urllib3.connectionpool:urlopen - Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:1129)'))': /api/1226033/envelope/ - pros-cli version:3.4.3
```

jman13378, 2023-09-09:
I was trying to upgrade the kernel but since the school doesn't have access to ssl i cant use pros nor git is there a was to get around the ssl issue like git has was to fix it

ayushuk, 2023-09-21:
At the moment we are looking to change our hosting away from github, I'll try to see if there is a way we can get around this temporarily.

ayushuk, 2023-09-21:
You seem to be on the latest kernel version so you wont need to upgrade but we will try to have a fix before the next kernel version update.

ayushuk, 2023-10-05:
Didn't mean to close

PR #296 (2024-01-18): ✨ Move PROS 4 to early access
#### Summary:
Deprecate PROS v3.

 - Renames beta to modern or v4
 - warns user v3 is deprecated once

Closes #295
closed 2024-01-18 by ayushuk