purduesigbots/pros-cli #331: Local Template Cannot Be Found in 3.5.0, able to 3.4.3

zxzimeng, 2024-02-09:
#### Expected Behavior:

`pros c fetch ARMS@3.2.0.zip`
`pros c apply ARMS@3.2.0`
successfully applies template to project

#### Actual Behavior:

`pros c fetch ARMS@3.2.0.zip`
sucessfull

`pros c query-templates`
does not include ARMS

`{'name': 'okapilib', 'version': '5.0.0', 'location': 'kernel-early-access-mainline', 'target': 'v5', 'local': True}
{'name': 'okapilib', 'version': '4.8.0', 'location': 'pros-mainline', 'target': 'v5', 'local': False}
{'name': 'okapilib', 'version': '4.7.0', 'location': 'pros-mainline', 'target': 'v5', 'local': False}
{'name': 'okapilib', 'version': '4.6.0', 'location': 'pros-mainline', 'target': 'v5', 'local': False}
{'name': 'okapilib', 'version': '4.5.0', 'location': 'pros-mainline', 'target': 'v5', 'local': False}
{'name': 'okapilib', 'version': '4.4.1', 'location': 'pros-mainline', 'target': 'v5', 'local': False}
{'name': 'okapilib', 'version': '4.4.0', 'location': 'pros-mainline', 'target': 'v5', 'local': False}
{'name': 'okapilib', 'version': '4.3.0', 'location': 'pros-mainline', 'target': 'v5', 'local': False}
{'name': 'okapilib', 'version': '4.2.0', 'location': 'pros-mainline', 'target': 'v5', 'local': False}
{'name': 'okapilib', 'version': '4.1.0', 'location': 'pros-mainline', 'target': 'v5', 'local': False}
{'name': 'okapilib', 'version': '4.0.5', 'location': 'pros-mainline', 'target': 'v5', 'local': False}
{'name': 'okapilib', 'version': '4.0.4', 'location': 'pros-mainline', 'target': 'v5', 'local': False}
{'name': 'okapilib', 'version': '4.0.3', 'location': 'pros-mainline', 'target': 'v5', 'local': False}
{'name': 'okapilib', 'version': '4.0.2', 'location': 'pros-mainline', 'target': 'v5', 'local': False}
{'name': 'okapilib', 'version': '4.0.1', 'location': 'pros-mainline', 'target': 'v5', 'local': False}
{'name': 'liblvgl', 'version': '8.3.7', 'location': 'kernel-beta-mainline', 'target': 'v5', 'local': False}
{'name': 'liblvgl', 'version': '8.3.6', 'location': 'kernel-early-access-mainline', 'target': 'v5', 'local': True}
{'name': 'liblvgl', 'version': '8.3.3', 'location': 'kernel-beta-mainline', 'target': 'v5', 'local': False}
{'name': 'liblvgl', 'version': '8.3.2', 'location': 'kernel-beta-mainline', 'target': 'v5', 'local': False}
{'name': 'liblvgl', 'version': '8.3.0', 'location': 'kernel-beta-mainline', 'target': 'v5', 'local': False}
{'name': 'liblvgl', 'version': '5.3.3', 'location': 'kernel-beta-mainline', 'target': 'v5', 'local': False}
{'name': 'kernel', 'version': '4.0.6', 'location': 'kernel-early-access-mainline', 'target': 'v5', 'local': True}
{'name': 'kernel', 'version': '4.0.5', 'location': 'kernel-beta-mainline', 'target': 'v5', 'local': False}
{'name': 'kernel', 'version': '4.0.4', 'location': 'kernel-beta-mainline', 'target': 'v5', 'local': False}
{'name': 'kernel', 'version': '4.0.3', 'location': 'kernel-beta-mainline', 'target': 'v5', 'local': False}
{'name': 'kernel', 'version': '4.0.0', 'location': 'kernel-beta-mainline', 'target': 'v5', 'local': False}
{'name': 'kernel', 'version': '3.8.0', 'location': 'pros-mainline', 'target': 'v5', 'local': False}
{'name': 'kernel', 'version': '3.7.3', 'location': 'pros-mainline', 'target': 'v5', 'local': False}
{'name': 'kernel', 'version': '3.7.2', 'location': 'pros-mainline', 'target': 'v5', 'local': False}
{'name': 'kernel', 'version': '3.7.1', 'location': 'pros-mainline', 'target': 'v5', 'local': False}
{'name': 'kernel', 'version': '3.7.0', 'location': 'pros-mainline', 'target': 'v5', 'local': False}
{'name': 'kernel', 'version': '3.6.2', 'location': 'pros-mainline', 'target': 'v5', 'local': False}
{'name': 'kernel', 'version': '3.6.0', 'location': 'pros-mainline', 'target': 'v5', 'local': False}
{'name': 'kernel', 'version': '3.5.4', 'location': 'pros-mainline', 'target': 'v5', 'local': False}
{'name': 'kernel', 'version': '3.5.3', 'location': 'pros-mainline', 'target': 'v5', 'local': False}
{'name': 'kernel', 'version': '3.5.2', 'location': 'pros-mainline', 'target': 'v5', 'local': False}
{'name': 'kernel', 'version': '2.12.2', 'location': 'pros-mainline', 'target': 'cortex', 'local': False}`

`pros c apply ARMS@3.2.0` 
raises
`pros.conductor.project.template_resolution.InvalidTemplateException: Could not find a template satisfying ARMS@>=0 (BaseTemplate) for v5`

downgrades to pros-cli==3.4.3

after fetching,
`pros c query-templates`
gives
`{'name': 'okapilib', 'version': '5.0.0', 'location': 'kernel-early-access-mainline', 'target': 'v5', 'local': False}
{'name': 'okapilib', 'version': '4.8.0', 'location': 'pros-mainline', 'target': 'v5', 'local': True}
{'name': 'okapilib', 'version': '4.7.0', 'location': 'pros-mainline', 'target': 'v5', 'local': False}
{'name': 'okapilib', 'version': '4.6.0', 'location': 'pros-mainline', 'target': 'v5', 'local': False}
{'name': 'okapilib', 'version': '4.5.0', 'location': 'pros-mainline', 'target': 'v5', 'local': False}
{'name': 'okapilib', 'version': '4.4.1', 'location': 'pros-mainline', 'target': 'v5', 'local': False}
{'name': 'okapilib', 'version': '4.4.0', 'location': 'pros-mainline', 'target': 'v5', 'local': False}
{'name': 'okapilib', 'version': '4.3.0', 'location': 'pros-mainline', 'target': 'v5', 'local': False}
{'name': 'okapilib', 'version': '4.2.0', 'location': 'pros-mainline', 'target': 'v5', 'local': False}
[... 40 lines cut ...]
install `pros-cli` with `pip install pros-cli==3.5.0`
download ARMS@3.2.0.zip from [here](https://github.com/purduesigbots/ARMS/releases/tag/v3.2.0)
`pros c new-project ./auton v5 3.8.0`
`mv` ARMS@3.2.0.zip to project folder
`cd` project folder
in project folder, `pros c fetch ARMS@3.2.0.zip`
in project folder, `pros c apply ARMS@3.2.0`
this line fails

`pip uninstall pros-cli`
`pip install pros-cli==3.4.3`
`pros c fetch ARMS@3.2.0.zip`
`pros c apply ARMS@3.2.0`
this works now

#### System information:
Operating System:  
OSX 14.3
python==3.10.13

PROS Version: 
pros, version 3.5.0
#### Additional Information

#### Screenshots/Output Dumps/Stack Traces

ayushuk, 2024-02-09:
Thanks for the report. Did you happen to use the early access flag at all before this? We have an attempted fix in #321 .

zxzimeng, 2024-02-09:
Im not sure what you meant by adding the "early access flag," so attached is fresh install of pros==3.5.0, using --disable-early-access and --early-access flags when building the new project, sorry if i misunderstood something

both still fail

`--disable-early-access` flag
`usr@ofzddRT567PCmbrBGgKKoYNbUjWHEVSv7f99xfgUJZUcPWcPSAerhx7E3if2h ~ % pros c new-project  ./test v5 --disable-early-access 
PROS 4 is now in early access. If you would like to use it, use the --early-access flag.
Updating kernel-beta-mainline... Done
Updating pros-mainline... Done
Downloading kernel@4.0.6 [####################################] 100%         
Extracting kernel@4.0.6 [####################################] 100%
Fetched kernel@4.0.6 from kernel-beta-mainline depot
Adding kernel@4.0.6 to registry...Done
Applying kernel@4.0.6 [####################################] 100%
Finished applying kernel@4.0.6 to test
Downloading okapilib@5.0.0 [####################################] 100%         
Extracting okapilib@5.0.0 [####################################] 100%
Fetched okapilib@5.0.0 from kernel-beta-mainline depot
Adding okapilib@5.0.0 to registry...Done
Applying okapilib@5.0.0 [####################################] 100%
Finished applying okapilib@5.0.0 to test
New PROS Project was created:
PROS Project for v5 at: /Users/usr/test (test)
Name      Version    Origin
--------  ---------  --------------------
kernel    4.0.6      kernel-beta-mainline
okapilib  5.0.0      kernel-beta-mainline
Building project...
Creating cold package with libc,libm,libpros,okapilib [OK]
Stripping cold package  [DONE]
Section sizes:
   text	   data	    bss	  total	    hex	filename
1155818	   3339	49433264	50592421	303faa5	bin/cold.package.elf
Compiled src/main.cpp [OK]
Adding timestamp [OK]
Linking hot project with ./bin/cold.package.elf and libc,libm,libpros,okapilib [OK]
Section sizes:
   text	   data	    bss	  total	    hex	filename
   1393	      4	48248834	48250231	2e03d77	bin/hot.package.elf
Creating cold package binary for VEX EDR V5 [DONE]
Creating bin/hot.package.bin for VEX EDR V5 [DONE]
usr@ofzddRT567PCmbrBGgKKoYNbUjWHEVSv7f99xfgUJZUcPWcPSAerhx7E3if2h ~ % pros c fetch         
usr@ofzddRT567PCmbrBGgKKoYNbUjWHEVSv7f99xfgUJZUcPWcPSAerhx7E3if2h ~ % cd test
usr@ofzddRT567PCmbrBGgKKoYNbUjWHEVSv7f99xfgUJZUcPWcPSAerhx7E3if2h test % pros c fetch LemLib@0.4.7.zip 
Extracting LemLib@0.4.7.zip  [####################################]  100%
Fetched LemLib@0.4.7 from local depot
Adding LemLib@0.4.7 to registry...Done
usr@ofzddRT567PCmbrBGgKKoYNbUjWHEVSv7f99xfgUJZUcPWcPSAerhx7E3if2h test % pros c apply LemLib@0.4.7
Updating kernel-beta-mainline... Done
Updating pros-mainline... Done
ERROR - pros.cli.main:main - Could not find a template satisfying LemLib@0.4.7 (BaseTemplate) for v5 - pros-cli version:3.5.0
  File "/Users/usr/.pyenv/versions/3.10.13/lib/python3.10/site-packages/pros/conductor/conductor.py", line 231, in apply_template
    raise dont_send(
pros.conductor.project.template_resolution.InvalidTemplateException: Could not find a template satisfying LemLib@0.4.7 (BaseTemplate) for v5
usr@ofzddRT567PCmbrBGgKKoYNbUjWHEVSv7f99xfgUJZUcPWcPSAerhx7E3if2h test % cd
`

`--early-access` flag
`usr@ofzddRT567PCmbrBGgKKoYNbUjWHEVSv7f99xfgUJZUcPWcPSAerhx7E3if2h ~ % pros c new-project ./testearlyaccess v5 --early-access
Early access is enabled. Using PROS 4.
Updating kernel-early-access-mainline... Done
Applying kernel@4.0.6 [####################################] 100%
Finished applying kernel@4.0.6 to testearlyaccess
Downloading liblvgl@8.3.7 [####################################] 100%         
Extracting liblvgl@8.3.7 [####################################] 100%
Fetched liblvgl@8.3.7 from kernel-beta-mainline depot
Adding liblvgl@8.3.7 to registry...Done
Applying liblvgl@8.3.7 [####################################] 100%
Finished applying liblvgl@8.3.7 to testearlyaccess
Applying okapilib@5.0.0 [####################################] 100%
[... 51 lines cut ...]
{'name': 'kernel', 'version': '4.0.6', 'location': 'kernel-early-access-mainline', 'target': 'v5', 'local': True}
{'name': 'kernel', 'version': '4.0.5', 'location': 'kernel-beta-mainline', 'target': 'v5', 'local': False}
{'name': 'kernel', 'version': '4.0.4', 'location': 'kernel-beta-mainline', 'target': 'v5', 'local': False}
{'name': 'kernel', 'version': '4.0.3', 'location': 'kernel-beta-mainline', 'target': 'v5', 'local': False}
{'name': 'kernel', 'version': '4.0.0', 'location': 'kernel-beta-mainline', 'target': 'v5', 'local': False}
{'name': 'kernel', 'version': '3.8.0', 'location': 'pros-mainline', 'target': 'v5', 'local': False}
{'name': 'kernel', 'version': '3.7.3', 'location': 'pros-mainline', 'target': 'v5', 'local': False}
{'name': 'kernel', 'version': '3.7.2', 'location': 'pros-mainline', 'target': 'v5', 'local': False}
{'name': 'kernel', 'version': '3.7.1', 'location': 'pros-mainline', 'target': 'v5', 'local': False}
{'name': 'kernel', 'version': '3.7.0', 'location': 'pros-mainline', 'target': 'v5', 'local': False}
{'name': 'kernel', 'version': '3.6.2', 'location': 'pros-mainline', 'target': 'v5', 'local': False}
{'name': 'kernel', 'version': '3.6.0', 'location': 'pros-mainline', 'target': 'v5', 'local': False}
{'name': 'kernel', 'version': '3.5.4', 'location': 'pros-mainline', 'target': 'v5', 'local': False}
{'name': 'kernel', 'version': '3.5.3', 'location': 'pros-mainline', 'target': 'v5', 'local': False}
[... cut ...]
ERROR - pros.cli.main:main - Could not find a template satisfying LemLib@0.4.7 (BaseTemplate) for v5 - pros-cli version:3.5.0
  File "/Users/usr/.pyenv/versions/3.10.13/lib/python3.10/site-packages/pros/conductor/conductor.py", line 231, in apply_template
    raise dont_send(
pros.conductor.project.template_resolution.InvalidTemplateException: Could not find a template satisfying LemLib@0.4.7 (BaseTemplate) for v5
usr@ofzddRT567PCmbrBGgKKoYNbUjWHEVSv7f99xfgUJZUcPWcPSAerhx7E3if2h testearlyaccess % 
`

BennyBot, 2024-02-09:
I believe this bug to be addressed in #321. If possible, could you test this for us? Reproduction of the bug has proven difficult on our end.

Steps:
1) disable whatever installation of pros-cli you currently have
2) download/clone the develop branch of this repository (purduesigbots/pros-cli)
3) With python 3.10 installed, in the root directory of this repository run `pip install -e ./`
4) ensure that the python Scripts folder is on your PATH (so the executable can be used in terminal)
5) make sure that `pros` command works. 

6) retry your above test case.

zxzimeng, 2024-02-09:
```usr@ofzddRT567PCmbrBGgKKoYNbUjWHEVSv7f99xfgUJZUcPWcPSAerhx7E3if2h ~ % python -V
Python 3.10.13
usr@ofzddRT567PCmbrBGgKKoYNbUjWHEVSv7f99xfgUJZUcPWcPSAerhx7E3if2h ~ % where python
/Users/usr/.pyenv/shims/python
usr@ofzddRT567PCmbrBGgKKoYNbUjWHEVSv7f99xfgUJZUcPWcPSAerhx7E3if2h ~ % cat .zshrc
export PYENV_ROOT="$HOME/.pyenv"
[[ -d $PYENV_ROOT/bin ]] && export PATH="$PYENV_ROOT/bin:$PATH"
eval "$(pyenv init -)"
usr@ofzddRT567PCmbrBGgKKoYNbUjWHEVSv7f99xfgUJZUcPWcPSAerhx7E3if2h ~ % gh repo clone purduesigbots/pros-cli
Cloning into 'pros-cli'...
remote: Enumerating objects: 6989, done.
remote: Counting objects: 100% (1211/1211), done.
remote: Compressing objects: 100% (576/576), done.
remote: Total 6989 (delta 751), reused 974 (delta 619), pack-reused 5778
Receiving objects: 100% (6989/6989), 28.07 MiB | 2.09 MiB/s, done.
Resolving deltas: 100% (4126/4126), done.
usr@ofzddRT567PCmbrBGgKKoYNbUjWHEVSv7f99xfgUJZUcPWcPSAerhx7E3if2h ~ % pip install -e pros-cli/       
Obtaining file:///Users/usr/pros-cli
  Preparing metadata (setup.py) ... done
Requirement already satisfied: click>=8 in ./.pyenv/versions/3.10.13/lib/python3.10/site-packages (from pros-cli==3.5.0) (8.1.7)
Requirement already satisfied: rich-click in ./.pyenv/versions/3.10.13/lib/python3.10/site-packages (from pros-cli==3.5.0) (1.7.3)
Requirement already satisfied: pyserial in ./.pyenv/versions/3.10.13/lib/python3.10/site-packages (from pros-cli==3.5.0) (3.5)
Requirement already satisfied: cachetools in ./.pyenv/versions/3.10.13/lib/python3.10/site-packages (from pros-cli==3.5.0) (5.3.2)
Requirement already satisfied: requests in ./.pyenv/versions/3.10.13/lib/python3.10/site-packages (from pros-cli==3.5.0) (2.31.0)
Requirement already satisfied: requests-futures in ./.pyenv/versions/3.10.13/lib/python3.10/site-packages (from pros-cli==3.5.0) (1.0.1)
Requirement already satisfied: tabulate in ./.pyenv/versions/3.10.13/lib/python3.10/site-packages (from pros-cli==3.5.0) (0.9.0)
Requirement already satisfied: jsonpickle in ./.pyenv/versions/3.10.13/lib/python3.10/site-packages (from pros-cli==3.5.0) (3.0.2)
Requirement already satisfied: semantic_version in ./.pyenv/versions/3.10.13/lib/python3.10/site-packages (from pros-cli==3.5.0) (2.10.0)
Requirement already satisfied: colorama in ./.pyenv/versions/3.10.13/lib/python3.10/site-packages (from pros-cli==3.5.0) (0.4.6)
Requirement already satisfied: pyzmq in ./.pyenv/versions/3.10.13/lib/python3.10/site-packages (from pros-cli==3.5.0) (25.1.2)
Requirement already satisfied: cobs in ./.pyenv/versions/3.10.13/lib/python3.10/site-packages (from pros-cli==3.5.0) (1.2.1)
Requirement already satisfied: scan-build==2.0.13 in ./.pyenv/versions/3.10.13/lib/python3.10/site-packages (from pros-cli==3.5.0) (2.0.13)
Requirement already satisfied: sentry-sdk in ./.pyenv/versions/3.10.13/lib/python3.10/site-packages (from pros-cli==3.5.0) (1.40.2)
Requirement already satisfied: observable in ./.pyenv/versions/3.10.13/lib/python3.10/site-packages (from pros-cli==3.5.0) (1.0.3)
Requirement already satisfied: pypng==0.0.20 in ./.pyenv/versions/3.10.13/lib/python3.10/site-packages (from pros-cli==3.5.0) (0.0.20)
Requirement already satisfied: pyinstaller in ./.pyenv/versions/3.10.13/lib/python3.10/site-packages (from pros-cli==3.5.0) (6.3.0)
Requirement already satisfied: typing in ./.pyenv/versions/3.10.13/lib/python3.10/site-packages (from scan-build==2.0.13->pros-cli==3.5.0) (3.7.4.3)
Requirement already satisfied: setuptools>=42.0.0 in ./.pyenv/versions/3.10.13/lib/python3.10/site-packages (from pyinstaller->pros-cli==3.5.0) (65.5.0)
Requirement already satisfied: altgraph in ./.pyenv/versions/3.10.13/lib/python3.10/site-packages (from pyinstaller->pros-cli==3.5.0) (0.17.4)
Requirement already satisfied: pyinstaller-hooks-contrib>=2021.4 in ./.pyenv/versions/3.10.13/lib/python3.10/site-packages (from pyinstaller->pros-cli==3.5.0) (2024.0)
Requirement already satisfied: packaging>=22.0 in ./.pyenv/versions/3.10.13/lib/python3.10/site-packages (from pyinstaller->pros-cli==3.5.0) (23.2)
Requirement already satisfied: macholib>=1.8 in ./.pyenv/versions/3.10.13/lib/python3.10/site-packages (from pyinstaller->pros-cli==3.5.0) (1.16.3)
Requirement already satisfied: charset-normalizer<4,>=2 in ./.pyenv/versions/3.10.13/lib/python3.10/site-packages (from requests->pros-cli==3.5.0) (3.3.2)
Requirement already satisfied: idna<4,>=2.5 in ./.pyenv/versions/3.10.13/lib/python3.10/site-packages (from requests->pros-cli==3.5.0) (3.6)
Requirement already satisfied: urllib3<3,>=1.21.1 in ./.pyenv/versions/3.10.13/lib/python3.10/site-packages (from requests->pros-cli==3.5.0) (2.2.0)
Requirement already satisfied: certifi>=2017.4.17 in ./.pyenv/versions/3.10.13/lib/python3.10/site-packages (from requests->pros-cli==3.5.0) (2023.11.17)
Requirement already satisfied: rich>=10.7.0 in ./.pyenv/versions/3.10.13/lib/python3.10/site-packages (from rich-click->pros-cli==3.5.0) (13.7.0)
Requirement already satisfied: typing-extensions in ./.pyenv/versions/3.10.13/lib/python3.10/site-packages (from rich-click->pros-cli==3.5.0) (4.9.0)
Requirement already satisfied: markdown-it-py>=2.2.0 in ./.pyenv/versions/3.10.13/lib/python3.10/site-packages (from rich>=10.7.0->rich-click->pros-cli==3.5.0) (3.0.0)
Requirement already satisfied: pygments<3.0.0,>=2.13.0 in ./.pyenv/versions/3.10.13/lib/python3.10/site-packages (from rich>=10.7.0->rich-click->pros-cli==3.5.0) (2.17.2)
[... cut ...]
  File "/Users/usr/.pyenv/versions/3.10.13/lib/python3.10/shutil.py", line 557, in copytree
    with os.scandir(src) as itr:
FileNotFoundError: [Errno 2] No such file or directory: ''
We detected something went wrong! Do you want to send a report? [y/N]: N
Not sending bug report.
Sentry is attempting to send 1 pending events
Waiting up to 2 seconds
Press Ctrl-C to quit
```

furthest i could get without intervention, has this to do with my environment? I have added the .zshrc contents, and the python version is gloabaly added to PATH

i will try without the venv

ayushuk, 2024-02-23:
Update: This still seems to be an issue. Download with curl gives an invalid zip file but even with a valid zip it seems we are unable to resolve the template. 
[image]