purduesigbots/pros-cli #342: 🐛Malformed Depot Breaks All CLI Functionality

meisZWFLZ, 2024-04-05:
#### Expected Behavior:

If I attempt to add a malformed depot, it shouldn't other commands.

#### Actual Behavior:

The malformed depot gets added to `conductor.pros`. This means that when you attempt to run any command, pros-cli will first attempt to parse `conductor.pros`, but since the depot was malformed, so is `conductor.pros`, and thus `jsonpickle` throws an error.

[error](https://github.com/purduesigbots/pros-cli/files/14877780/err.log)

#### Steps to reproduce:

With npm installed:
powershell:
```powershell
mkdir depot;
'[{"py/object":"pros.conductor.templates.base_template.BaseTemplate","metadata":{},"name":""}]' | Out-File -FilePath ./depot/depot.json ascii;
Start-Process npx -ArgumentList "-y http-server depot -p 8080";
sleep 6;
pros c add-depot malformed http://127.0.0.1:8080/depot.json;
pros c query-templates --force-refresh some-text;
pros --version
```
not windows:
```bash
mkdir depot 
echo '[{"py/object":"pros.conductor.templates.base_template.BaseTemplate","metadata":{},"name":""}]' >./depot/depot.json 
npx -y http-server depot -p 8080 & 
sleep 6
pros c add-depot malformed http://127.0.0.1:8080/depot.json
pros c query-templates --force-refresh some-text
pros --version
```
This also happens attempting to apply any template.

If you want to fix your cli after running the above commands, remove the malformed depot from `conductor.pros`.

#### System information:
Operating System:  Windows 11
pros-cli: 3.5.1

ayushuk, 2024-04-05:
Sorry for this being an issue. We'll take a look. In the meantime, if your CLI breaks due to a malformed `conductor.pros` use `pros c reset` to reset the malformed file.

meisZWFLZ, 2024-04-05:
`pros c reset` throws the same error as `pros --version` with a broken `conductor.pros`. But its fine, since I already manually fixed the `conductor.pros` file.