purduesigbots/pros-cli #369: ✨Remove Kernel from Project

SizzinSeal, 2024-07-11:
#### Requested Feature

Ability to remove the PROS kernel from a PROS project

#### Is this Feature Related to a Problem?

It is not possible to make a PROS template that is 100% independent of the PROS kernel

#### Benefits of Feature

Some templates could be completely independent of PROS, such as [LemLib/units](https://github.com/LemLib/units). However the PROS CLI does not allow you to remove the PROS kernel from a project. This is necessary if one still wishes to use the PROS conductor for applying templates.

It would also make it possible for the PROS kernel to depend on PROS templates. For example: solving [purduesigbots/pros #680](https://github.com/purduesigbots/pros/issues/680) by adding [LemLib/units](https://github.com/LemLib/units) as a dependency

#### Additional Information

This may need some modification of [purduesigbots/pros](https://github.com/purduesigbots/pros),  but since it was the CLI that threw the error I opened a feature request here.

The error message in question:
```
> pros c uninstall kernel        
Uninstalling kernel@4.1.0
ERROR - __main__:main - Cannot remove the kernel template. Maybe create a new project? - pros-cli version:3.5.4
  File "pros\conductor\conductor.py", line 352, in remove_template
  File "pros\conductor\project\__init__.py", line 170, in remove_template
ValueError: Cannot remove the kernel template. Maybe create a new project?
Sentry is attempting to send 1 pending events
Waiting up to 2 seconds
Press Ctrl-Break to quit
```

BennyBot, 2024-07-11:
I don’t think making `pros c uninstall kernel` work directly is the best answer. I would rather not deal with people saying “my pros disappeared from my pros” (you know how vtow is, they always find a way).

It should be possible to achieve what you are trying by manually removing files, editing files, etc (either before or after the pros m template command, not sure which), but I agree this is not what you should have to do.

I would be fine with adding a new command for converting a project into a kernel independent project (i.e. removing kernel from the project with a separate command so that users don't accidently do it). It may also be easier to change `pros make template` to have an additional kernel dependence flag, eg. `pros make template --no-kernel` (this is probably what I would want implemented).

In regards to LemLib units being a default library (or, as you suggested, a dependency for the kernel), I really believe the only default templates in a project should be kernel and liblvgl. As far as kernel depending on LemLib units (and my opinion on issue [#680](https://github.com/purduesigbots/pros/issues/680) as a whole), I don't think it is a kernel level change, at the most it would be a first party template. Even so, I don't want anybody to be forced to use a library or set of units. If someone wants to develop their own units library (no matter how bad it might be), they should still be able to do that, without having another units library in their project. In my mind, making the kernel depend on LemLib units would be essentially forcing users to use LemLib units, as opposed to having the opportunity to develop their own units. I'm not a lead anymore though, so final decision will rest with @AndrewLuGit and @phinc23 for what to do.

Obviously, we want to **strongly suggest** a units library like yours to our users (high/top spot on the branchline page, many other template devs will use it as a dependency, etc.), so I do encourage you to keep working on LemLib units. Your team will probably be the first to hear about any branchline developments as we prepare for alpha stage (no timeline on that as we are all very busy this summer, eg. I have 2 jobs and classes so do not have time to commit to this yet).

SizzinSeal, 2024-07-11:
Whether LemLib/units ends up being a dependency to the PROS kernel is not relevant to this discussion, I merely used it as an example. I intend to pursue it further in the purduesigbots/pros kernel as well as through discussion on social platforms.

BennyBot, 2024-07-11:
Yes, I agree. In regards to the original issue with cli template generation, something should be done about that. We can continue discussion on the units library on the [#680](https://github.com/purduesigbots/pros/issues/680) page or on discord.

ion098, 2024-11-20:
@SizzinSeal now that #379 is merged, I think this can be closed.

closed 2024-11-22 by SizzinSeal