purduesigbots/pros-cli #370: ✨PROS Template Packs

SizzinSeal, 2024-07-15:
#### Requested Feature
PROS template packs. Apply multiple PROS templates with a single zip file

#### Is this Feature Related to a Problem?
Modular templates (such as LemLib) may depend on other templates. Currently, users either have to install all the dependencies manually, which is time consuming and can lead to user errors.

#### Benefits of Feature
Modular templates such as LemLib can depend on other templates without complicating the install process. This would be made obsolete with branchline, but until branchline is released, this feature should be "good enough"

#### Additional Information
Why does LemLib now depend on other templates?
Because LemLib has been split into multiple smaller templates, and is developing new separate templates that have common dependencies, such as logging and units.

BennyBot, 2024-07-15:
This is already a planned feature in branchline.

SizzinSeal, 2024-07-15:
This is proposed as a temporary solution to a problem while we wait for branchline to be released

SizzinSeal, 2024-07-15:
After discussion with @BennyBot, LemLib will write its own install script while we wait for branchline to be released.

closed 2024-07-15 by SizzinSeal

BennyBot, 2024-07-15:
For clarification on the discussion, changes involving conductor will likely not occur until after the completion/release of branchline.