purduesigbots/pros-cli #391: 🐛Won't update code.

sroar365, 2025-03-14:
#### Expected Behavior:

To download code into robot.
#### Actual Behavior:

Claims it download but did nothing at all. (Sometimes)
#### Steps to reproduce:

Yes. It just happens.
#### System information:
Operating System:  
Windows 11 Build 24.H
PROS Version: 

#### Additional Information

#### Screenshots/Output Dumps/Stack Traces

BennyBot, 2025-03-14:
it would be wonderful if you could provide screenshots of this. Are you saving and rebuilding your code before you upload? Can you show confirmation that your code built successfully?

sroar365, 2025-03-15:
We have already resolved the issue. It was because we did not save and rebuild the code. Thanks though.

closed 2025-03-15 by mayankpatibandla