purduesigbots/pros-cli #392: ✨Detect library on board before uploading

vrobot-2, 2025-05-11:
#### Requested Feature

Detect whether the library is on board before compressing and uploading. 

#### Is this Feature Related to a Problem?

#### Benefits of Feature

This should reduce upload times by more than 2x.

#### Additional Information

Are there existing/further optimizations to be made for upload out of curiosity?

BennyBot, 2025-05-11:
We already do this? If the cold package changes we have to reupload the whole thing, as the cold package contains every library (kernel, liblvgl, whatever else you use etc.). If it doesn’t change, we only upload the hot package (stuff in src folder ie opcontrol, autonomous, initialize, any files you made, etc. ) If you are talking about something like:

User has kernel and liblvgl uploaded already 
User adds lemlib
User reuploads code, and now only needs to upload lemlib

I’m not sure if this is possible.

vrobot-2, 2025-05-24:
I'm entirely sure what happens but it says compressing library and then is like oh the library is already on board. Couldn't it check before hand before it compresses it? I'm just speculating I'm not really sure. 👍 thanks for the explanation.

edjubuh, 2025-05-24:
> Couldn't it check before hand before it compresses it?

Unfortunately, no. The way we know the library is uploaded is by comparing a checksum of what's on your PC to what's on the device. The checksum is performed on the compressed library.

https://github.com/purduesigbots/pros-cli/blob/1185b7d144fa54790ec7c86e85c51783371da91b/pros/serial/devices/vex/v5_device.py#L475

You could disable compression. If the libraries are larger, this comes with trade offs with longer upload times when something changes and increased storage usage.

> User reuploads code, and now only needs to upload lemlib

If my memory serves, you can only link to one library (VEX design choice). If, from PROS perspective, you want to use multiple libraries, PROS combine them into one larger library and use the combined result.